JakobEngel/dso #221: Is this a valid test setup using OpenCV?

Fredrum, 2020-06-21:
Hi,

since the Pangolin example wasn't working for me on my Raspberry Pi 4 i'm trying to get this running using some rough opencv test code.  Is there anyone here that would be able to tell me if the following should work?

(some of the code above this taken from the ros example)
```
    cv::Mat frame;
	ImageAndExposure * fImg = new ImageAndExposure(640, 480, 0);
	int id = 0;
	while(1)
	{
		// wait for a new frame from camera and store it into 'frame'
        cap.read(frame);
        // check if we succeeded
        if (frame.empty()) {
            printf("ERROR! blank frame grabbed\n");
            break;
        }
        // show live and wait for a key with timeout long enough to show images
        cv::cvtColor(frame, frame, cv::COLOR_BGR2GRAY);
        cv::imshow("Live", frame);
        // Press  ESC on keyboard to  exit
		char c = (char)cv::waitKey(1);
		if( c == 27 ) 
			break;
		
		// convert image and feed dso
		float* tmp = (float*)frame.data;
		*fImg->image = *tmp; // copy the data, not the pointer itself
		fullSystem->addActiveFrame( fImg , id);

		id += 1;
	} 
```

Cheers
Fred

tanjunyao7, 2020-07-01:
hi, I think
`*fImg->image = *tmp; // copy the data, not the pointer itself`
only copies the first pixel value,not the whole image. Maybe 
`memcpy(fImg->image, tmp, sizeof(float)*frame.rows*frame.cols);`
Besides, the image should be undistorted before fed to fullsystem.

Fredrum, 2020-07-01:
Thank you for the advice, I will get back to this and do what you say!
Cheers!