JakobEngel/dso #237: No timestamps in result file

dgkntrz, 2021-04-11:
Hello,

When I run the algorithm on EuRoC dataset, the timestamps in results.txt are all 0. Any ideas on how to solve this problem?

Thanks in advance!

moajouz, 2024-07-12:
hello @QsingHuan I am facing this problem currently too, I am providing the times.txt file and a .zip folder containing my dataset. This did not also give me any results. should I provide the location in ```inline void loadTimestamps(/path/times.txt)``` as I also tried it on sequence11 from TUM dataset and also got me no results. this is how I am calling it using terminal: 
```-files=/home/mooo/aub/datasets/TUM/sequence_11/images.zip \
-calib=/home/mooo/aub/datasets/TUM/sequence_11/camera.txt \
-vocabPath=/home/mooo/HSLAM/misc/orbvoc.dbow3 \
-mode=1 ```

ashwinjoseph95, 2024-09-23:
when i try placing tiems.txt in dataset folder i get:
```
cv::imread could not read image /home/knnas/dso_ws/DATASET/2011_10_03/2011_10_03/2011_10_03_drive_0027_sync/image_02/data_resized/times.txt! this may segfault. 
Segmentation fault (core dumped)
```
Apparantly the program thinks the times.txt is an image file