JakobEngel/dso #242: pangolin don't work during build dso

qian2020, 2022-02-20:
[image]

sarossilli, 2022-02-22:
I recently was able to build DSO.

Try rebuilding/installing pangolin? It looks like it cannot find the pangolin headers.

qian2020, 2022-02-23:
I used the pangolin v6.0 + Ubuntu 18.04. Maybe there is something wrong to build DSO?

qian2020, 2022-02-23:
> I recently was able to build DSO.
> 
> Try rebuilding/installing pangolin? It looks like it cannot find the pangolin headers.

I used the pangolin v6.0 + Ubuntu 18.04. Maybe there is something wrong to build DSO?

vanguard478, 2022-04-08:
I think in Pangolin v0.6 there was a major refactoring. Try removing the Pangolin v0.6 (in the build directory do `sudo make uninstall` and installing v0.5 from https://github.com/stevenlovegrove/Pangolin/releases/tag/v0.5

donaldduckkkkk, 2024-02-23:
you dont install your pangolin. 
`cd ~/pangolin_ws`  && mkdir build &&  cd build && cmake .. && make && sudo make install