JakobEngel/dso #248: errors occur when make FullSystem.cpp

zhangletian2, 2022-04-24:
the errors are as follows:
make[2]: *** [CMakeFiles/dso.dir/build.make:63: CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/dso.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

and here are the specific errors(_1 _2 _3 are also not declared) 
/home/happy/dso/src/FullSystem/FullSystem.cpp:646:117: error: ‘_4’ was not declared in this scope
646 | ointsMT_Reductor, this, &optimized, &toOptimize, _1, _2, _3, _4), 0, toOptimize.size(), 50);
      |

Matthew-Bradley, 2022-05-05:
I'm experiencing the same problem. Does anyone know when the last time was when DSO compiled sucessfully?

lukasvst, 2022-05-05:
I think the reason is the newer Boost version.
You could try adding `using namespace boost::placeholders;` to line 36 of `dso/src/util/IndexThreadReduce.h`

That's what I did to make DM-VIO work with new Boost versions.

Matthew-Bradley, 2022-05-05:
That did the trick. I also had to swap:

CV_LOAD_IMAGE_GRAYSCALE with cv::IMREAD_GRAYSCALE
CV_LOAD_IMAGE_COLOR with cv::IMREAD_COLOR0
CV_LOAD_IMAGE_UNCHANGED cv::IMREAD_UNCHANGED

in dso/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp:

minz27, 2023-03-05:
> 

Just pointing out a small typo, it's cv::IMREAD_COLOR not cv::IMREAD_COLOR0. Thanks for posting the solution though, it works for me now