JakobEngel/dso #259: undefined reference to `boost::system::system_category()' and multiple other errors!

kurosh-z, 2022-12-05:
Hi, 
make -j8 results in multiple errors:

`
kurosh@ubuntu:~/repos/dso/build$ make -j8
Scanning dependencies of target dso
[  4%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp.o
[  8%] Linking CXX static library lib/libdso.a
[ 92%] Built target dso
Scanning dependencies of target dso_dataset
[ 96%] Building CXX object CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o
[100%] Linking CXX executable bin/dso_dataset
/usr/bin/ld: warning: libOpenNI.so.0, needed by /usr/local/lib/libpangolin.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libOpenNI2.so.0, needed by /usr/local/lib/libpangolin.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: in function `_GLOBAL__sub_I__Z8vignetteB5cxx11':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: in function `boost::thread_exception::thread_exception(int, char const*)':
/usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: /usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: /usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: /usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o:/usr/local/include/boost/thread/exceptions.hpp:51: more undefined references to `boost::system::system_category()' follow
/usr/bin/ld: lib/libdso.a(FullSystem.cpp.o): in function `_GLOBAL__sub_I__ZN3dso12FrameHessian15instanceCounterE':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(FullSystem.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(FullSystemOptimize.cpp.o): in function `boost::condition_error::condition_error(int, char const*)':
/usr/local/include/boost/thread/exceptions.hpp:84: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(FullSystemOptimize.cpp.o): in function `boost::thread_exception::thread_exception(int, char const*)':
/usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: /usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: /usr/local/include/boost/thread/exceptions.hpp:51: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(FullSystemOptimize.cpp.o):/usr/local/include/boost/thread/exceptions.hpp:51: more undefined references to `boost::system::system_category()' follow
/usr/bin/ld: lib/libdso.a(FullSystemOptimize.cpp.o): in function `_GLOBAL__sub_I__ZN3dso10FullSystem21linearizeAll_ReductorEbPSt6vectorIPNS_18PointFrameResidualESaIS3_EEiiPN5Eigen6MatrixIdLi10ELi1ELi0ELi10ELi1EEEi':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(FullSystemOptimize.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(FullSystemOptPoint.cpp.o): in function `_GLOBAL__sub_I__ZN3dso10FullSystem21optimizeImmaturePointEPNS_13ImmaturePointEiPNS_30ImmaturePointTemporaryResidualE':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(FullSystemOptPoint.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(FullSystemDebugStuff.cpp.o): in function `_GLOBAL__sub_I__ZN3dso10FullSystem17debugPlotTrackingEv':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(FullSystemDebugStuff.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(FullSystemMarginalize.cpp.o): in function `_GLOBAL__sub_I__ZN3dso10FullSystem28flagFramesForMarginalizationEPNS_12FrameHessianE':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(FullSystemMarginalize.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(Residuals.cpp.o): in function `_GLOBAL__sub_I__ZN3dso18PointFrameResidual15instanceCounterE':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(Residuals.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(CoarseTracker.cpp.o): in function `_GLOBAL__sub_I__ZN3dso13CoarseTrackerC2Eii':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(CoarseTracker.cpp.o): in function `__static_initialization_and_destruction_0':
/usr/local/include/boost/system/error_code.hpp:222: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: /usr/local/include/boost/system/error_code.hpp:223: undefined reference to `boost::system::system_category()'
/usr/bin/ld: lib/libdso.a(CoarseInitializer.cpp.o): in function `_GLOBAL__sub_I__ZN3dso17CoarseInitializerC2Eii':
/usr/local/include/boost/system/error_code.hpp:221: undefined reference to `boost::system::generic_category()'
/usr/bin/ld: lib/libdso.a(CoarseInitializer.cpp.o): in function `__static_initialization_and_destruction_0':
[... 84 lines cut ...]
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnGetDepthMap'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnContextUnregisterFromShutdown'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `oniStreamGetSensorInfo'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnGetIRMap'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `oniDeviceInvoke'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnOSFreeAligned'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `oniGetDeviceList'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `oniStreamIsPropertySupported'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnCreateImageGenerator'
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnFrameSyncWith'
[... cut ...]
/usr/bin/ld: /usr/local/lib/libpangolin.so: undefined reference to `xnSetMapOutputMode'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dso_dataset.dir/build.make:161: bin/dso_dataset] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/dso_dataset.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`
There were also some other errors with ‘CV_LOAD_IMAGE_COLOR’ being undefined and I changed it to cv::IMREAD_COLOR .
I ran the make on Ubuntu 20.04 inside the virtual machine and OpenCV version 4.2.0. 
Any suggestion on what could be causing these errors?

kurosh-z, 2022-12-06:
UPDATE: I installed boost v. 1.67.0. It seems to be happy now with the boost::system but I'm still getting an error at 88% :
 
 `
kurosh@ubuntu:~/repos/dso/build$ make -j8
Scanning dependencies of target dso
[  4%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptimize.cpp.o
[  8%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemOptPoint.cpp.o
[ 16%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemMarginalize.cpp.o
[ 16%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o
[ 20%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/FullSystemDebugStuff.cpp.o
[ 24%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/CoarseTracker.cpp.o
[ 32%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/CoarseInitializer.cpp.o
[ 32%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/Residuals.cpp.o
[ 36%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/ImmaturePoint.cpp.o
[ 40%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/HessianBlocks.cpp.o
[ 44%] Building CXX object CMakeFiles/dso.dir/src/FullSystem/PixelSelector2.cpp.o
[ 48%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/EnergyFunctional.cpp.o
[ 52%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/AccumulatedTopHessian.cpp.o
[ 56%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/AccumulatedSCHessian.cpp.o
[ 60%] Building CXX object CMakeFiles/dso.dir/src/OptimizationBackend/EnergyFunctionalStructs.cpp.o
[ 64%] Building CXX object CMakeFiles/dso.dir/src/util/settings.cpp.o
[ 68%] Building CXX object CMakeFiles/dso.dir/src/util/Undistort.cpp.o
[ 72%] Building CXX object CMakeFiles/dso.dir/src/util/globalCalib.cpp.o
[ 76%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/OpenCV/ImageDisplay_OpenCV.cpp.o
[ 80%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp.o
[ 84%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/Pangolin/KeyFrameDisplay.cpp.o
[ 88%] Building CXX object CMakeFiles/dso.dir/src/IOWrapper/Pangolin/PangolinDSOViewer.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/dso.dir/build.make:206: CMakeFiles/dso.dir/src/OptimizationBackend/EnergyFunctional.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/dso.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

IldarGreat, 2022-12-07:
Hi, you can build dso with help Readme from https://github.com/IldarGreat/dso